GrazerComputerClub/WiringPi #27: improvement for gpio wfi possible?

Joerg12345, 2024-05-23:
Thanks for the work translating WiringPi for the Pi5 !

Is it possible that the command `gpio ... wfi ...` returns the status of the GPIO (0 or 1) when the interrupt occurs ?

For example, I use `gpio -g wfi <gpioNumber> both`.
After the interrupt occurs, I then have to re-read the GPIO with the command `gpio -g read <gpioNumber>` to find out whether the interrupt occurred at `falling (0)` or `rising (1)`.
If gpio wfi returned the status at which the interrupt occurred, I would no longer have to re-read the GPIO.

Thank you!
Jörg

mstroh76, 2024-07-17:
Yes its possible and useful 
I had the same problem, but with the Library ISR Function
The new ISR reports the level after IRQ, but it is not implemented so far. 
So we need new/advanced ISR implementation, new API call, new parameter (wifs) for gpio tool.

mstroh76, 2024-07-17:
Issue moved to  https://github.com/WiringPi/WiringPi/issues/264

closed 2024-07-17 by mstroh76