GrazerComputerClub/WiringPi #29: Operating System

MageMCU, 2024-09-19:
This is more of a question than the issue that revolves around the operating system(s).  The WiringPi Library installation can be confusing for a beginner for both the hardware and its OS. Over the years, I have heard Debian, Raspbian and Raspberry Pi OS. Presently I use the latter and the question or suggestion, could you please clarify and maybe specifically expand on your installation guide for both source and pre-built. I use the older and newer pi hardwares... When I install your lib, the geany IDE has an error with the include directives <wiringpi>, or <wiringPi.h> etc... Thank you - MageMCU Github

5shreya, 2024-09-30:
Hey I would like to work on this issue. Can you assign me

MageMCU, 2024-10-01:
Hey what’s up GCC WiringPi

I remember writing something at your github site… I just wrote a new article for beginners but also a study for myself. 

RaspberryPi-Uno-Communication (https://github.com/MageMCU/RaspberryPi-Uno-Communication)
The title of the article: 1013 Pi Uno SMBus I2C Study

I would appreciate to hear from you about the article…

Jesse Carpenter
carpentersoftware.com
MageMCU Github

MageMCU, 2024-10-04:
Hey what’s up

The issue is located at WiringPi Github which by the way is not my account and I am not a contributor there either. I am presently working at MageMCU Github. Although it may have some issues, I am presently not looking for contributors.

Can you give me some information about yourself and your back ground and expertise. Maybe base on what you reveal, I can find you a spot so you can participate at MageMCU but if you do not hear from me then I cannot help you…

Jesse

virtuosonic, 2024-10-23:
Hi MageMCU, what error message do you get? It looks like this lib's header by default are installed to /usr/local/include when building from source, the deb package installs to /usr/include

MageMCU, 2024-10-23:
I was referring to the GrazerComputerClub (also includes the WiringPi/WiringPi GitHub account) - WiringPi - ReadMe file. This couild be rewritten for beginners to grasp as to how to setup their environment which includes both the hardware and its OS where I personally use Raspberry Pi and its OS.

Personally I have no issue with WiringPi

Please expand on your WiringPi - ReadMe file.

Thanks

MageMCU